billroy/bitlash #42: Cannot use backspace in terminal

mondalaci, 2015-01-19:
It'd be very nice to be able to use backspace in the terminal. Currently nothing happens upon pressing backspace. I guess bitlash should echo backspace towards the client and shorten the command buffer by one character.

- A Pinoccio user :)

matthijskooijman, 2015-01-19:
@mondalaci, I actually think what you suggest is implemented already, both in regular bitlash as well as in the Pinoccio version (which uses its own input-processing loop, here: https://github.com/Pinoccio/library-pinoccio/blob/master/src/Shell.cpp#L2696-L2705).

Could you clarify what code you're running exactly?

Perhaps your terminal isn't sending regular backspace characters? What terminal are you using?

mondalaci, 2015-01-19:
@matthijskooijman I'm using GNU screen on Linux and having the latest firmware that has been fetched by HQ.

matthijskooijman, 2015-01-19:
Hmm, seems this is something specific with screen - it doesn't work for me either when using screen. I suspect screen is sending different bytes for backspace, I'll have a closer look soon. In the meanwhile, you could try using a different terminal (I use minicom).

mondalaci, 2015-01-19:
Minicom indeed works! I'd also be interested to figure out what's going on
with screen and how can I make it work properly in this respect. Thanks!

matthijskooijman, 2015-01-19:
Turns out that screen sends a delete (0x7f) instead of backspace. I think this is a common problem with terminals, I remember BSD vs Linux to also suffer from problems like this. I've modified the pinoccio code to [handle both as backspace](https://github.com/Pinoccio/library-pinoccio/pull/226). It turns out that [bitlash already does this as well](https://github.com/billroy/bitlash/blob/master/src/bitlash-cmdline.c#L165), should have looked better at bill's code when writing our own :-)

In any case, I think this ticket can be closed.

mondalaci, 2015-01-19:
@matthijskooijman Mad props to you for figuring out and fixing legacy bullshit like this! :)

Closing ticket.

closed 2015-01-19 by mondalaci

billroy, 2015-01-20:
Thanks for following up, Matthijs.

FYI/FWIW, here's what I use to avoid the issues with screen and minicom.  I wrote a little terminal emulator named "bloader.js" that you can install via npm.  Source here: https://github.com/billroy/bloader.js

ticpu, 2015-01-20:
For a CLI tool there is also picocom which has less options and a couple
useful shortcuts. In between screen and minicom.