billroy/bitlash #43: Talk to arduino without >command

borgdrone7, 2015-03-13:
Hi,
I am trying to use bitlash with C#. Everything is perfect and works except one thing which ruins everything for me. Bilash by default thinks that I am in terminal and echos my commands and > sign. However in order for it to be usable when talking with C# program over serial it must send only result of the command. Currently when I write printf("a0:%d", a0) as a response I will get:
printf("a0:%d", a0)
a0:551
>
I want to get:
a0:551

I am sure I am missing something obvious but I cannot seem to find a solution. Whats the catch?

Thanks in advance!

billroy, 2015-03-13:
Hello and thanks for your note.

You can modify Bitlash so it doesn’t echo characters (though this will make it more difficult to use manually, of course).

In src/bitlash-cmdline.c at line 117 you will find the “spb(c)” that echoes the input character back over the serial link.  Comment that function call out and you should be good to go.

Happy to take followup questions if that doesn’t work for you.

-br

borgdrone7, 2015-03-13:
Hi,
Thank you very much for quick answer. Commenting that line solved echo, however there is still "> " which "pollutes" the output. How can I remove that too. I just want result of the command. Another question is, is it possible to upload function but that it is not automatically saved to EPROM. That way I can define all function in C# strings and every time connection is started just upload them with no need for EPROM, so I could have more functions.
Thanks again for making so wonderful library. I looked at all libraries out there and I think your's is by far the best.

billroy, 2015-03-13:
You can suppress the prompt by defining an empty function named “prompt”:

	function prompt {}

Your question about functions “not saved to EEPROM” puzzles me.  Where would you suggest to store them?  It would be possible to store them in RAM, I suppose, on the larger-ram Arduinos.

-br

borgdrone7, 2015-03-13:
Hi,
Yes, Arduino Mega has 8Kb of SRAM and 4Kb of eprom. As for prompt, I figured it out on my own and came here to let you know, and saw your answer :). I am getting one extra newline in output, maybe it is due to my code on client side. Do you send extra newline together with response sometimes or it is always just one to end the response? I am sending "print a0" and sometimes I get "\n" first and then "531\n" and sometimes I get immediately "531\n".

billroy, 2015-03-14:
It may help to comment out the call to speol() at line 150 in src/bitlash_cmdline.c - this is generated when /r or /n is entered.

The print command normally generates one newline at the end of the output unless you end the print command with a comma.

The printf() function gives you complete control over whether or not there’s a newline at the end.

-br

borgdrone7, 2015-03-14:
I will try it (speol) thanks.