billroy/bitlash #44: run bitlash code stored in flash

pwl42, 2015-03-20:
hi Bill,
  have you ever built a version that would interpret bitlash code that is stored in the unused part of flash (on a 32K mcu)?   it seems to me that would be a good place to put a larger program.

billroy, 2015-03-21:
Hello and thanks for your question.

You can define functions in flash by adding them to the “builtin_table” data structure in the file src/bitlash-builtins.c; see the doc there for the use of the BUILT_IN macro.  Functions built-in this way can be run just like eeprom functions.

I hope that helps, and I’m happy to take followup questions.

-br

pwl42, 2015-03-21:
I guess I am just being too picky.   I would prefer to load unused flash memory blocks with bitlash code and have it executed the same way as code in eeprom.  Those blocks could be replaced without recompiling.  Of course, this would be useful only on the larger devices.

On the internet nobody can hear you being subtle.   --Linus Torvalds

--------------------------------------------

 

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 it on GitHub.

billroy, 2015-03-21:
Thanks for clarifying.  If I understand correctly, you would like to write the flash dynamically at runtime, like the eeprom.

Last time I checked, it was considered tricky to write to flash at runtime, and there were no libraries available to do it.

If, however, someone has developed a library to write to flash at runtime, this would not be a difficult feature to implement.

-br

pwl42, 2015-03-21:
I was thinking that I would write to those unused blocks when I loaded bitlash and If I needed to change them later I would use avrdude.  Is the flash readable only as 16 bit words and, if so, is that a problem?

On the internet nobody can hear you being subtle.   --Linus Torvalds

--------------------------------------------

 

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 it on GitHub.

billroy, 2015-03-21:
IIRC the flash is only writable in pages of some size considerably greater than one byte.  And I believe there is some complication about having to be in bootloader mode to write to it, though I may be recalling that incorrectly.

-br

closed 2015-03-24 by pwl42