billroy/bitlash #45: eeprom overflow error with bitlash run command for a user defined function

eshanMewantha, 2015-04-24:
Hello, I am using an arduino mega. I have the following code. Which works as expected.

```
#include <bitlash.h>

void setup(void) {
    initBitlash(9600);
}

int count =1;

void loop(void) {
  runBitlash();
  
  if (count ==1){
    doCommand("{pinMode(13, 1); dw(13,1);  if(a0< 200){print \"level=\"a0;}; snooze(2000); }");
    //doCommand("function iot {pinMode(13, 1); dw(13,1);  if(a0< 200){print \"level=\"a0;}; snooze(2000); }");
    //doCommand("run iot")
    count =0;  
  }  
  delay(1000);   
}
```

When I try to run it as a function in background using;
``` 
if (count ==1){
   doCommand("function iot {pinMode(13, 1); dw(13,1);  if(a0< 200){print \"level=\"a0;}; snooze(2000); }");
   doCommand("run iot;")
   count =0;  
  }
```

even though the first doCommand should return 'saved', it returns 'eeprom overflow'. I checked the eeprom using eeprom read and write and also used eeprom clear. Everything is fine. (Plugged out and connected and all the other programs works fine, so does the arduino doCommands. Error occurs only when I try to run a function in background. I have used "ps;" to check whether there are anyother background functions running. There are none.) 

I should also mention that the background run command and saving worked perfectly for few attempts. Then it started giving the 'eeprom overflow' instead of 'saved'. 

Am I missing something here ? 
Any insight will be much appreciated. Thank You

eshanMewantha, 2015-04-24:
Update:
I was able to Temporarily fix the problem by writing the entire eeprom with eeprome_clear using the value 255 instead of the default 0. It seems that when doCommand tries to save a function to eeprom, it looks for the allocated positions but instead of the normal norm of thinking 0 as unused memory, it searches for 255. Thus cleaning the eeprom and writing it with 0 (as the usual practice) wont help. Even though cleaning and filing the eeprom with 255 before loading the program works, it still causes some error after few uploads. To get rid of it I again have to clean the eeprom with setting 255 to all the locations.

billroy, 2015-04-24:
Hello and thanks for your notes.

As you have observed, the AVR convention is that 255 is the value for uninitialized EEPROM.

The fact that the problem recurs after several reboots from the clean state leads me to believe what you are seeing is likely caused by eeprom fragmentation.  Could you send the output of “peep” after the error happens?

A possible workaround is to only define the eeprom function if it is not already there.  You could possibly use the function src/bitlash-eeprom.c::findKey(char *id) to check if it’s already been defined from your C driver program.

-br

eshanMewantha, 2015-04-27:
Hello, Thank you for the reply.

When everything is working fine the peep of the eeprom is this.

```
E000:  iot_ func  tion $pin  Mode (13,   1);  dw(  13,1 );    if(a 0< 2  00){ prin  t "l evel
E040:  ="a0 ;};   snoo ze(2  000) ;$..  .... ....  .... ....  .... ....  .... ....  .... ....
E080:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E0C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E100:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E140:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E180:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E1C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E200:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E240:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E280:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E2C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E300:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E340:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E380:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E3C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E400:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E440:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E480:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E4C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E500:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E540:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E580:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E5C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E600:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E640:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E680:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E6C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E700:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E740:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E780:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E7C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E800:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E840:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E880:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E8C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E900:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E940:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E980:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E9C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
EA00:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
EA40:  .... ....  .�� ���� ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���
EA80:  �������  ��� ���� ��� ���  ��� ���  ��� ���  �������  ��� ���� ��� ���
EAC0:  �������  ������  ��� ���� ��� ���  ��� ���  �������  ��� ���� ��� ����
EB00:  ��� ���  �������  ��� ���� ��� ���  ��� ���  ��� ���  �������  ��� ����
EB40:  ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���  �������  �������
EB80:  ��� ���  ��� ���  �������  ��� ���� ��� ���� ��� ���  ��� ���  �������
EBC0:  ��� ���  ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���  ��� ���
EC00:  ��� ���� ��� ���  ��� ���  �������  ������  ��� ���� ��� ���  ��� ���
EC40:  ��� ���� ��� ���� ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���
EC80:  �������  ��� ���� ��� ���  ��� ���  ��� ���  �������  ��� ���� ��� ���
ECC0:  �������  ������  ��� ���� ��� ���  ��� ���  �������  ��� ���� ��� ����
ED00:  ��� ���  �������  ��� ���� ��� ���  ��� ���  ��� ���  �������  ��� ����
ED40:  ��� ���  �������  �������  ��� ���� ��� ���  ��� ���  �������  ������
ED80:  ��� ���  ��� ���  �������  ��� ���� ��� ���� ��� ���  ��� ���  �������
EDC0:  ��� ���  ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���  ��� ���
EE00:  ��� ���� ��� ���  ��� ���  �������  ��� ���  ��� ���� ��� ���  ��� ���
EE40:  ��� ���� ��� ��� ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���
EE80:  �������  ��� ���� ��� ���  ��� ���  �������  �������  ��� ���� ��� ���
EEC0:  �������  ��� ���  ��� ���� ��� ���  ��� ���  �������  ��� ���� ��� ����
[... cut ...]
EEC0:  ��� ���  �������  ��� ���� ��� ���  ��� ���  �������  �������  ��� ����
EF00:  ��� ���  �������  ��� ���  ��� ���� ��� ���  ��� ���  �������  ��� ����
EF40:  ��� ���  ��� ���  �������  ��� ���� ��� ���  ��� ���  ��� ���  �������
EF80:  ��� ���  ��� ���  �������  �������  ��� ���� ��� ���  ��� ���  �������
EFC0:  ��� ���� ��� ���  ��� ���  �������  ��� ���� ��� ��� ��� ���  ��� ���
```

I think for the time being I would try out the workaround you mentioned. 
Thank you again for the reply.