billroy/bitlash #46: Blank user function causes following functions not to appear

felicitus, 2015-06-15:
When one accidentally declares an empty function, additional functions don't show up in the list.

```
> function foo {}
saved
> function bar { print "foo"; }
saved
> ls
function foo {};
function  print "foo";  {};
```

billroy, 2015-06-16:
Hi, Timo:

Thanks for your note.  You’ve found a good little bug here.  I’m not clear whether it’s a bug in cmd_ls() or whether the function definition is broken, but something is definitely wrong.  Unfortulately, I’m away from my workbench this week so I can’t debug it first-hand.  Can you please send the results of a “peep” at the eeprom after you create that situation?

-br

felicitus, 2015-09-19:
Here's the dump:

```
E000:  .... ....  .... ....  .... ....  .... ....  .... ....  ..fo o$$b  ar$  prin  t "f oo";
E040:   $.. ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
```

It doesn't seem to be the cmd_ls() function, because I'm not able to call the function "bar" in that example:

```
> function foo {}
saved
> function bar { print "foo"; }
saved
> bar()

-----^
unexpected number
> ls
function foo {};
function  print "foo";  {};
> rm foo
> ls
function bar { print "foo"; };
> bar
foo

```

As soon as I remove the empty function `foo`, the call to bar works again.