billroy/bitlash #47: Port to spark-core or esp-8266

creationix, 2015-06-15:
I would love to use bitlash on my various wifi-enabled development boards.  They all have a port of arduino's wireing which I imagine is most the work porting bitlash, but I'm wondering what else would be required and if anyone has tried this before.

For example, when trying to import the bitlash header into an ESP8266 project using the Arduino IDE, I get errors about bitlash trying to include avr.h.

I'd be willing to do the work to port what's left, but I'm looking for some guidance.

billroy, 2015-06-16:
Hi, Tim, and thanks for your note.

If you already have a wiring port available, most of the work will be sorting out the header file includes to be correct for your platform.  As you’ve already found, avr.h is included, but you don’t want that.

Have a look in src/bitlash.h and you will see several chunks of code which configure for different platforms.  You’ll need to do something similar.  You’ll probably need to add an #ifdef for your platform and include the wiring header files.  You’ll also need to #ifdef out the avr stuff.

I hope that helps.  Let me know if you hit a snag.

Kind regards,

-br

igrr, 2015-07-24:
Hi Tim, did you get any further? It would be cool if we could make Bitlash ESP8266-compatible. Let me know if you need some help (here or in esp8266 gitter chat).
Thanks.

creationix, 2015-07-24:
bitlash makes a lot of avr and arduino assumptions.  I'm currently working on a language kinda like bitlash, but developing for esp chips first.  I have one based on esp8266/Arduino and am starting one using esp-open-sdk directly. https://github.com/creationix/uscript

q2dg, 2015-08-20:
Maybe ESP8266 and Spark go a bit out of scope of Bitlash but I think official ARM-based Arduino boards (Due and Zero at the moment) should be default-supported. 
Thanks anyway.

tochinet, 2015-08-21:
OTOH, on ARM/ESP8266 boards, you also have Espruino as an alternative. This
gives you complete Javascript capabilities, much more than what bitlash
intended to do.

creationix, 2015-08-21:
@tochinet well yes, and esp8266 in particular has a popular elua port called nodemcu, but it's not the same category as bitlash.  I'm looking for a simpler language.

ioeverything, 2016-04-22:
If you wana experience micropython in esp8266 you can find something in 
https://github.com/micropython
http://docs.micropython.org/en/latest/esp8266/index.html