billroy/bitlash #48: Not working on new Arduino..

Curnelius, 2016-06-02:
On the new boards using ARM processors, like the Zero, which are the future of MCU's, you get an error on compilation : 

> fatal error: avr/io.h: No such file or directory
>  #include "avr/io.h"
                     

Wonder if its possible at all ?

creationix, 2016-06-02:
Back when I looked at porting bitlash to the esp8266 I found that it makes a lot of assumptions about avr and arduino. It's a very simple language, it might make sense to re-implement it for other platforms.  If we're going to do that then it might be time to re-evaluate the language decisions in light of the more powerful 32-bit processors.

This is an area in which I'm doing active research and would love to collaborate with someone.

Curnelius, 2016-06-02:
Thats right. I would like to read more about the design patterns required to write it again like a "pro", do you have something in mined ?

q2dg, 2016-07-25:
Oooh, I would be really great if there was an platform-agnostic Bitlash. Really really great. 
On the other hand, last commit in this github was more than two years ago...so maybe it's time to reboot.

creationix, 2016-07-25:
@q2dg what exactly about bitlash (as compared to JS, lua, ruby, python, forth, lisp) do you like?  Is is just the idea of scripting on a microcontroller in general or do you like the specific design of bitlash?

q2dg, 2016-07-25:
The idea of real-time interaction with the microcontroller as if I was in a telnet session (with the possibility of creating own scripts to run on it, yes) . Thanks.

creationix, 2016-07-25:
@billroy Would you be opposed to a new language being implemented that borrows heavily from the design of bitlash?

billroy, 2016-07-25:
Quite the contrary - I would encourage forks and forward experimentation in new directions!

-br

q2dg, 2016-07-25:
I haven't the technical skill to opinate about it. I'd only want front-end syntax were similar to Arduino's one.

creationix, 2016-07-25:
Arduino is just C++ within a framework/library.  It's not the easiest syntax to parse, but it's certainly the most popular style (JavaScript/Java/C/C++/C#/PHP/etc are all C-like syntax)