billroy/bitlash #49: Can't save user functions to startup?

Curnelius, 2016-07-26:
I hope I am getting it wrong, but after trying many times, I could define a new user function and fire it, but I could not set the system to fire it on startup, no matter what .

So in Arduino : 

`  addBitlashFunction("ran", (bitlash_function) ran);
`

and the function 

```
numvar ran(void) {

//also print not working .
 digitalWrite(13, HIGH);   // delay only for this matter
  delay(1000);                
  digitalWrite(13, LOW);     
  delay(1000);  

}
```

When you try to run it , it works, but if you try to 

`function startup { ran; }`  // here you get back "saved"

you get on boot the error - **"unexpected number"** , no matter what you do in this function.

I hope I am doing it wrong, otherwise whats the benefit of user functions ? 

Thanks anyway!

billroy, 2016-07-26:
Does your sketch register the bitlash user function by calling addBitlashFunction?

It would be something like: addBitlashFunction(“ran", (bitlash_function) ran);

-br

Curnelius, 2016-07-26:
Its right there inside my question, please read **line 4**. And as I said, the function is running if you just type it, and working, but you can't add it to memory for startup, no matter what you do, on startup you get an error of unexpected number.

billroy, 2016-07-26:
Sorry, I overlooked that.

What you are doing should work, I believe.  

I wonder about the state of your eeprom.  Does “ls” or “peep” show anything unusual?

-br

Curnelius, 2016-07-26:
ok , lets say I call the function **"my function"** , I register it as I said, and I can type it and see how it works.

Then I do this : 
`function startup { my function;
`}

if I run ls i get : 
```
> ls
function startup { my function;}
```;

if i run peep i get : 

```
E000:  star tup$   myf unct  ion; $...  .... ....  .... ....  .... ....  .... ....  .... ....
E040:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E080:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E0C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E100:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E140:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E180:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E1C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E200:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E240:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E280:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E2C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E300:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E340:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E380:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....
E3C0:  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ....  .... ...
```

if i boot i get : (the function is not fired there is an error )
```
bitlash here! v2.0 (c) 2013 Bill Roy -type HELP- 944 bytes free
unexpected number
> 
```

Curnelius, 2016-07-26:
I guess you are not going to update it anyway  , so we will have to leave it without this functionality. 
(wish I knew how to add it quickly enough)

billroy, 2016-07-26:
Stumped here.  Are you able to post the complete text of the actual sketch file you’re using?

-br

Curnelius, 2016-07-26:
Is this serious? I just showed you exactly, twice , what we do, you could try that yourself its your software, it takes 1 minute. Even with your examples of the Tone, you can't save this function to memory no matter what you do. it makes to whole thing much less appealing. (sorry for saying that but I'v spent too much reading all docs and so )

anyway , arduino : 

```
#include "bitlash.h"

void setup(void) {

	 	initBitlash(9600); 
   addBitlashFunction("myfunction", (bitlash_function) my function)

}

void loop(void) {
	runBitlash();
}

numvar myfunction(void) {

//also print not working .
 digitalWrite(13, HIGH);   // delay only for this matter
  delay(1000);                
  digitalWrite(13, LOW);     
  delay(1000); 
   
    return 0;

}
```

Command line (9600) :  

`> my function
`

works , then : 

```
> function startup { myfunction; }
saved
> ls
function startup { myfunction; };
> boot
��������here! v2.0 (c) 2013 Bill Roy -type HELP- 944 bytes free
unexpected number
```

creationix, 2016-07-26:
> Is this serious? I just showed you exactly, twice , what we do, you could try that yourself its your software, it takes 1 minute.

That's some mighty fine entitlement there.  I'm sorry you're frustrated, but unless you're paying for support you have no right to demand even 1 minutes of someone's time.  

Also you didn't post the full source twice, you posted 1 line twice.  @billroy ignored your rude attitude and simply asked for more information so he could help you.

creationix, 2016-07-26:
@Curnelius also note the email icon next to his posts.  That means he's replying via email, probably on his mobile and is not at a computer where we can test the code.

Curnelius, 2016-07-26:
@creationix I already paid, I paid in 1 work day that provided me nothing because I was relying on someone else's docs to do the basic things they should do.

In a website like this the rules should be different from "hey I only did you a favour" .

True , he does not have to spent a second on me, but with this approach of "This is my code, you can relay on what I said in the docs, but its your problem" , then the whole github thing is worthless.

One can say, of course, he is only contributing and thats really great , but this brings serious companies to never relay on this website, and thats much worse.

I did provide the exact line he was looking for in my comment , and then he asked something that made me feel nobody even cared to read what I wrote.

Anyway, I respect the work he has done, but I feel frustrated because the most basic thing is not working as what you would be expected from the docs.

creationix, 2016-07-26:
@Curnelius you spent time, that's not the same as paying a person.

Anyway, I'm sorry you're frustrated.  I'm not trying to scold you, I'm trying to help you understand that he is probably helping you best he can.  

If you continue being rude, he might stop helping you.  I've seen it happen so many times before on other projects.

Curnelius, 2016-07-26:
I think I'v provided more then enough information so that the owner of the code would be able to replicate the bug within 45 seconds and not letting me write it again and again.
If he is out of home he could say that.(as he did before)

I don't think I am rude . I only expect people to read. 
I will not die if he wouldn't help me- he stopped updating this code years ago anyway. There is Firmata and our custom solution.

Curnelius, 2016-07-26:
You know what, if I am rude I am so sorry. I had a pretty bad week. Building a company is hard.
So I am sorry . Anyway I will leave this because I don't think there is solution at all.

billroy, 2016-07-26:
Curnelius:

May I offer a couple of points to close out the thread?

1.  I appreciate your apology.  You have noted earlier in the thread that this project isn't getting a lot of time.  Threads like this are part of the reason.  I'm in this for the fun, and it is simply no fun to provide help when someone takes an attitude.  I hope you will think about how you might approach future requests for help with more light and less heat.  If it isn't fun, we're not doing it right.

2.  When asked for a sketch, please provide actual working code copied from your text editor.  There were two compiler errors in the code you sent which were easy to fix but which make it harder to help you because of the uncertainty about exactly what you were running there.

3.  It turns out your problem is easily resolved.  You have found an old Bitlash bug, for which there is a simple workaround.  As you discovered, a user function cannot be run directly from the startup function.  However, user functions can be run from eeprom functions without problems at any time other than startup, which is sort of special.  So the workaround is to add a 'trampoline function' for your startup function to call, something like this, where "myfunction" is the name of your user function:

    function runit { myfunction; }
    function startup { runit; }
    
Since the user-defined function is not run directly from the startup function, the bug is avoided.  This works here with the code you sent, amended to compile.  I'd be happy to know whether or not this sorts out your problem, and good luck with your project!

4. Tip of the hat to Tim:  Yeah, I was out of pocket.  Good eye.    

Back to coding, everyone.  And be careful out there.

-br

Curnelius, 2016-07-26:
Thanks for your respond, there is some gap between people who makes projects for fun and the ones who try to relay on them to make big things :)

Unfortunately , I have tried what you said right when I encounter the bug, I even tried many variations of it, and its not working. it also not working with your own examples of user function ( did you try this approach ?).

Anyway this project could lead to some good things, and I think you should had take to be more then just fun ;)

closed 2016-07-26 by billroy

Curnelius, 2016-07-26:
funny that you provided a solution that does not works and then close the discussion .
Hope you tried it yourself at least.

creationix, 2016-07-26:
@Curnelius he did test the code you gave him after fixing the typos. 

> This works here with the code you sent, amended to compile

Also as the creator of the ticket, you should be able to reopen it if the solution doesn't work.

> Anyway this project could lead to some good things, and I think you should had take to be more then just fun ;)

I agree that bitlash is a very interesting design.  I'm also designing scripting languages for microcontrollers (which is why I watch this repo).  I'd be willing to design, implement and maintain something if I had some guidance as to requirements/needs.

Do you need to support tiny systems like the arduino uno?  Or can you use more powerful systems like the 32-bit microcontrollers that can barely run elua or a real-time OS?

Curnelius, 2016-07-26:
Listen , you have to be serious here. His solution does not work ! a little bit of honesty guys.
Do you serious enough to try it, or you just copy what he says ?
My code compiles without any modification, and,  you can just check it your self and see. Whats wrong with you people? you really know how to write code? if you do, spend 45 seconds and see that its not working (!)  its not working even with the guy's own examples of "Tone" function.

I have tried any possible variation and its not working.

This discussion is really stupid, feels like 1 programmer speaking with 2 philosophers. Go and check things your self .

I guess its just a project he had to do for university , otherwise I can't see why he is so un-passionate about it , to ACTUALLY go and check whats wrong with it- something that will take me a week.

Yes , we would like to support both small but also large processors for some special mission. I think I will write the whole of it myself. 

This is why I can't never trust other people's classes that made just for "fun" , where my life rely on this.

creationix, 2016-07-26:
No I didn't try it, I don't have an uno.  But Bill said he did and I took his word for it.

If your original code compiles for you, but not for him, then something is different, more context is required to help you.  If his modified code works for him, but not for you, then again something else is different.  Perhaps some more details would help.

What board are you flashing, what IDE version are you using, what settings do you have etc?

Curnelius, 2016-07-27:
Thats a lie, I spent 3 hours on this, its not working, and if it did, the guy would show the full code.
He only wants to close this and feel good about himself. thats not ok here at all.

This is going no where, its pretty stupid, Good luck.

creationix, 2016-07-27:
@Curnelius great news!  I found an arduino mega in my parts drawer and can reproduce your problem.  I'm using the tone example from the git repo without any changes.  When I try to call it from startup I get `unexpected number`.  When I try the workaround proposed by Bill it still doesn't work.

Bad news is we're going to need Bill's help to track this down.  But knowing how little time he has for this to begin with and after you essentially called him a self-serving liar, I would not be surprised if he refused to help you.

From your point of view the library is not working as advertised and you're frustrated because you had hoped it would work for you.  This is very understandable, I've been there many times.  

But you must have no idea how draining it is to spend hundreds or sometimes thousands of hours creating something for free and giving it away for people to then make more demands and accuse you of not being a good person for years.  

I've also been in Bill's shoes many times and am trying to be the peace maker here.  I want bitlash to work for you and I want Bill to enjoy sharing his creation with the community.  We need more bright creative minds like his.

You say that you've spent 3 hours on this and therefore are entitled to make demands on someone's hobby project.  If you don't like it, don't use it.  Make your own.  It's not like Bill is forcing you to use his project and then suddenly is too busy to give you proper support.

If I was in your shoes I would politely ask Bill for support. Be clear and professional about my issues.  If he brushes me aside I might either move on or be persistent in explaining my issue. But above all, I would be patient because I know he's doing me a service by giving away his software and offering any level of free support.

If it turned out that Bill simply couldn't help me and I was unable to dig into the code myself, I would move on.  That's what professionals do.  They don't call people liars, they don't make character judgements.  You demand professional support while acting like a child.  That's not OK here at all.

(And actually this is the voice of experience.  I tried to use bitlash a while ago. I wanted esp8266 support.  It became clear that Bill didn't have the resources/interest to do that for me so I've since started designing my own language/runtime.)

Curnelius, 2016-07-27:
Well , the guy lied. He clearly said he tested it , and it works with my code, he also said my code is wrong. (Then deleted it ! )

Thats not moral.

If we dealing with lies here , thats a whole new issue and there is nothing more to talk about.

I don't want to work with it even if it turns out to be good.( and will not recommend it to others because of this guy )

Not cool.

Curnelius, 2016-07-27:
"This works **here** with the code you sent, amended to compile" .

Amazing guy. ('work here', like he is actually testing it right now, with my "wrong" code- that complies both on mac and pc, strange people....  ).

creationix, 2016-07-27:
Never assume that just because something works on your machine, it works on everyone's machine.

And never, ever assume someone is lying just because they say it's different on their machine.

boneskull, 2016-07-27:
it's weird, last Monday I put my old tv out on the curb with a sign reading "FREE". By Tuesday the tv was gone. But then Wednesday, @curnelius rings my doorbell and demands I fix one of the hdmi ports, which is a little wonky. I told him the hdmi ports worked fine for me, but then he calls me a liar and storms off.  He still had the tv

Curnelius, 2016-07-27:
Well , if github is a street with broken old tv's and garbage that doesnt work, and people who lies about them, then you right thats weird.

I would go to the Apple store from now on.

jaitaiwan, 2016-07-28:
Quality meme's right here 👍 

In all seriousness though one of the hardest parts of open source maintenance is this...

It's generally best that when things go sour early on, to drop the conversation and leave things as is and not progress the issue or allow further comments. Otherwise tensions increase, make people say things they normally wouldn't and ruin most everyone's reputation who gets involved. It's probably best that at the least this issue gets closed to collaborators only and even get the comments deleted. That way it saves face for everyone and we can go back to being civil human beings and hopefully not be judged by having our comments eternally saved in the github ether.

Just my unsolicited 2c 😛