billroy/bitlash #54: Serial flash support?

colesnicov, 2017-09-23:
Hello. And what about support for running a program from serial flash memory, for example from Winbond 25Q32BVSIG.
Thanks

billroy, 2017-09-25:
Thanks for your note.  It is a fascinating idea!  But I do not have time to contribute to it.  It is not a difficult problem once you have the flash memory driver sorted out.

-br

colesnicov, 2017-09-25:
Hello. Do you need to implement some interface or the existing library already? Personally, I want to try this one,  [Marzogh/SPIFlash](https://github.com/Marzogh/SPIFlash/wiki/Compatibility), as soon as I get the most mentioned flash chip. But otherwise for arduino there are a lot of libraries that allow communication with serial memory ...

Sorry for my bad english :)