billroy/bitlash #59: bitlash demo - blink does not blink

rmd06, 2019-02-23:
Hi, I followed the tutorial and run a blink startup function to make LED connected to D7 pin blink.

After I run startup, I found that the LED was always on, and no blink.

Then I run a command to modify the callee toggle7 to see if something changed:

function toggle7 {d7=0;} // I founs that LED was suddenly off
function toggle7 {d7=1;}// I founs that LED was suddenly on
function toggle7 {d7=!d7;} // I expect LEDto blink, but it failed.

rmd06, 2019-02-23:
I found that when I just define the fucntion:
function toggle7 {d7=!d7;}
The D7 LED suddenly turned on after define the function.

rmd06, 2019-02-24:
I found the reason:
The dr function or d7 always return 0 no matter the voltage is high or low.
perhaps before reading d7, the pinmode was converted to input, and the power was shutoff on d7, so the read value was always 0.

rmd06, 2019-02-24:
A better way is to use gobal variable a-z to make switch.
below is the matlab code anf it works well:

aStr = 'a=0;';
fprintf(obj1,'%s\n',aStr);
aStr = 'function toggle7 {pinmode(7,1);if(a==0){dw(7,1);a=1;}else{dw(7,0);a=0;}}';
fprintf(obj1,'%s\n',aStr);
pause(0.3)
aStr = 'function blink7 {run toggle7,1000;}';
fprintf(obj1,'%s\n',aStr);
pause(0.3)
aStr = 'blink7()';
fprintf(obj1,'%s\n',aStr);

rmd06, 2019-02-24:
I also found that when run blink in background, the LED intensity was not stable at high frequency,.
However, if I run it in foreground, LED is very stable.

aStr = 'i=0; while(i<1000){toggle7();i++}';
fprintf(obj1,'%s\n',aStr);

billroy, 2019-02-24:
On some Arduino boards, for example the Due, the Arduino digitalRead() function does not properly read back state values written to pins configured as output pins.  This works correctly on the original ‘328 Arduino.

Is it possible you are using a board other than a standard ‘328 Arduino?

-br

rmd06, 2019-02-24:
Hi, below is the arduino information cli shows:
% Arduino/Genuino Uno	arduino:avr:uno    
The devide manager shows it as "USB-Serial-CH340-COM3".
This is not the original uno, but one  bought from Taobao(very cheap, only $4).