billroy/bitlash #61: ========^ unexpected number error for functions

ilovetogetspamed, 2019-09-26:
Please Help!  I thought I was writing my functions properly.  Also, example "timer1" doesn't work either.   I'm using an Authentic Arduino Mega2560 R3.

### I/O
```
bitlash here! v2.0 (c) 2013 Bill Roy -type HELP- 6044 bytes free
> print needle()

--------------^
unexpected number
> print NEEDLE()

--------------^
unexpected number
> echo(100,1,"Spam, Spam Spam", 42)
100
1
Spam, Spam Spam
42
> print timer1(), timer1(), timer1()

--------------^
unexpected number
```

### Code
```
#if defined(ARDUINO) && ARDUINO >= 100
  #include "Arduino.h"
#else
  #include "WProgram.h"
#endif
#include "bitlash.h"
#include "../bitlash/src/bitlash.h"  // for sp() and printInteger()

numvar func_NEEDLE(void)
{
    return analogRead(A8);
}

// The ECHO() function echoes back its arguments for debugging, with proper handling for strings
numvar func_ECHO(void) {
  for (int i=1; i <= getarg(0); i++) {
    if (isstringarg(i))
      sp((const char *) getstringarg(i));
    else printInteger(getarg(i), 0, ' ');
      speol();
  }
}

numvar timer1(void) {
    return TCNT1; 	// return the value of Timer 1
}

void loadBitlashFunctions()
{
  addBitlashFunction("ECHO", (bitlash_function) func_ECHO);     // The ECHO() function echoes back its arguments, \
                                                                   with proper handling for strings
  addBitlashFunction("NEEDLE", (bitlash_function) func_NEEDLE);
  addBitlashFunction("timer1", (bitlash_function) timer1);
}

void setup(void) {
  initBitlash(57600);   // must be first to initialize serial port
  loadBitlashFunctions();
}

void loop(void) {
  runBitlash();
}
```

billroy, 2019-09-27:
Sorry for the delay in following up.  

I see you are using uppercase names for your custom user functions.  Bitlash internally compares names in all lower case.  What happens if you use lowercase function names in the addBitlashFunction calls?

This would not explain your issue with timer1, which I am still pondering.

Kind regards,

-br

ilovetogetspamed, 2019-09-29:
Billroy,

I changed to lower case.  That helped.

Chris

ilovetogetspamed, 2019-09-30:
changing my callback names to lower case worked.

ilovetogetspamed, 2019-09-30:
BTW, timer1 was not an issue.  I was using that as an 'extra' function call in my testing of your library.

closed 2019-09-30 by ilovetogetspamed

billroy, 2019-09-30:
Thanks for clearing up that loose end… I still didn’t have a theory on it.

Kind regards,

-br